Sensirion/arduino-uart-sfx6xxx #1: Feature request: working code with Arduino Opta

lnegrini, 2023-09-27:
We need to control SFC6000 mass flow controllers with PLC. To achieve this goal we are trying to program an Arduino Opta which has an STM32H747XI mcu with builtin Modbus TCP and serial RS485.

But we aren't able to communicate to mf controllers by setting the Ota serial port. More exactly we are  using the example and having a problem here:
```
// Adjust as needed for you Arduino board.
// [Serial, Serial1, Serial2, etc.]
#define SENSOR_SERIAL_INTERFACE Serial1

SensirionUartSfx6xxx sensor;
```

Do you have any suggestion how to solve the problem? Thanks

qfisch, 2023-09-29:
Hello @lnegrini ,

Unfortunately we do not have an Arduino Opta around here, and we currently have no plans to create a complete example using this module.
Looking at this page from Arduino : https://docs.arduino.cc/tutorials/opta/getting-started-with-rs485
My first intuition would be to import the RS library and replace all the transmission by the RS485.write.
Did you try a similar approach already ?

qfisch, 2023-12-22:
I will close that issue since it is inactive

closed 2023-12-22 by qfisch