Sensirion/arduino-uart-sfx6xxx #10: Troubleshooting UART connection

adamabr00, 2024-05-10:
Hi, 

This sensor looks really promising. I got the SFM6000D-20SLM. I am using an Arduino Giga R1. 

I continually encounter "Timeout while reading data" errors no matter what I try. I have also tried to find documentation for communicating with the sensor but can't find any guide for this. There seems to be a lot of data for I2C communication but not very much for UART. 

I'm at the verge of giving up but figured I'd give it a last shot by asking here. What would be a good way to troubleshoot the sensor? 

Thanks, 
Adam

LeonieFierz, 2024-05-13:
Hello Adam

thank you for contacting us and considering our sensor.

Can you please give some more information about the connection between the sensor and the Arduino Giga? 

According to the documentation on https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/#serialuart-pins there are 5 serial connections available. Please make sure the serial interface defined in the example to communicate with the sensor matches with the PINs you connect to on the Arduino.

`#define SENSOR_SERIAL_INTERFACE Serial1`   

If you use Serial1, then you need to use D0/D1 pins on the Arduino. Make also sure you have the right cable connected to  TX/RX (just try swapping them to make sure). 

Best, Léonie

adamabr00, 2024-05-13:
Hi Léonie

Thanks for coming back to me! Hopefully, we can figure this out together. 

This is the schematic for my connection:

[image]

To interface with the sensor I use a RS485 to TTL module. Link: [https://www.amazon.com/dp/B07B667STP?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

I use a 24 volt power supply with screw terminals to provide power to the sensor. 
I use an M8 cable to connect the sensor to the rest of the system. Link: [https://www.amazon.com/dp/B0CKPL996D?ref=ppx_yo2ov_dt_b_product_details&th=1](url)

Couple things I have noticed/tried: 

1. The RS485 to TTL module I used has two LEDs, one for when data is transmitted and one for when data is received.
2. I would see both LEDs blinking with everything connected.
3. When I disconnect the 24V power supply from the sensor I only see a one way transmission, suggesting that the sensor was sending data back. 
4. I have tried to swap both the A+ and B- wires as well as the RX and TX wires. It is only in one configuration that I see both LEDs blink on the RS485 to TTL module. However, all combinations result in the same error "Timeout while reading data". 
5. I have made sure the supplied power is within the range of the sensor. And the sensor and the Arduino share ground. 
6. I have tried different Serial ports. My general understanding is that Serial1 refers to (RX2, TX2) and Serial2 refers to (RX3, TX3) and so on. This was also confirmed with the LED's, where no data was sent or transmitted on other serial ports. 
7. It was difficult to know which of the cables in the M8 cable that corresponds to which pin, but the coloring of the cables comply with the suggested colors for the pinout. 

I have been wondering if there is any other way to see if the sensor is working, like an onboard LED or similar but have not found any.

LeonieFierz, 2024-05-14:
Hi Adam

thank you for the detailed Infos. It seems you tried out pretty much everything already. As it is always tricky to get everything plugged in correctly, I suggest some points you might have already tested.

- Why did you connect the +24V to the VIN of the Arduino? Is it to power the board? Could you power the Arduino over USB or the power jack
- In point 3. you mention that the LED suggest that the sensor is sending back data when is is disconnected from power. I would then only expect the Arduino to send out data, but not any data coming back from the sensor. Thus actually that you see the RXD LED on, but not the TXD when the sensor is not powered. If this is ok, then the Arduino connection to the board should work.
- Can you check again the wiring from the adapter to the sensor please. In your schematic you have the D- connected to A+ and D+ to B-. You should have it the other way around, D- to B- and D+ to A+. (only try swapping one side at a time)

adamabr00, 2024-05-20:
Hi Léonie

Sorry for my late response. I appreciate your help! 

- I connected the +24V to the Arduino to power the board. When coding and debugging I can power the board through USB but when the board is used in it's real application that is not possible. I don't see how this would make any difference. 
- I realize my writing was unclear in point 3. I meant that when the power to the sensor was removed, then no data transmission could be seen coming back to the Arduino, only the Arduino sending data. Before that, when the sensor was plugged in, I could clearly see both LEDs blinking. This suggested that the sensor was sending back data when it was connected to power. But somehow that still was not working with the Arduino. 
- I know I have tried switching D- and D+ connections without luck.

LeonieFierz, 2024-05-23:
Hi Adam

As I do not have the hardware here to test out your setup, it is hard to say where the problem is. The "Timeout while reading data" exception indicates that the sensor is not sending back any data, which is most likely the connection. The only thing I can advice is checking again through all the wiring setup. I tried to find out what the LED on your RS485 adapter actually indicate, but I did not find any details, so I not sure if having them blinking means that the communication is working. Make sure you do not use the Serial line also used for communication to the board over USB (  e.g. Serial4 using RX3 (Pin 15) / TX3 (Pin 14) should be fine).

Best, Léonie

psachs, 2024-07-23:
@adamabr00 do you have any update? Could get the code working?

adamabr00, 2024-07-23:
@psachs Unfortunately, I did not get it working. I had more pressing issues and this had to be set to the side.

LeonieFierz, 2024-08-20:
@adamabr00 please create a new issue or reopen this one once you get back to work on it

closed 2024-08-20 by LeonieFierz