Sensirion/arduino-uart-sfx6xxx #11: Troubleshooting Execution Errors

bposk, 2024-05-27:
Hi, 

I am currently trying to get a SFC6000D-5slm device working in some capacity. I have gone back and forth trying the I2C and UART libraries without much luck. The best implementation I have gotten so far is doing a UART communication from Arduino Mega <-> RS485 to TTL Converter <-> SFC6000D and uploading the arduino-uart-sfx6xxx exampleUsage code. 

When running the code I am unable to properly communicate with the sensor and am mostly met with a :
`Error trying to execute readAveragedMeasuredValue(): Execution error, status register: 0x42`
error. I am also unable to find any documentation on the Execution error itself. Has anyone dealt with this issue or have had difficulties getting the SFC6000D properly working? Any Advice helps

Thanks!

LeonieFierz, 2024-05-28:
Hi @bposk 

You can find the meaning of the device specific errors in https://sensirion.com/media/documents/88CA2961/65156AEC/GF_AN_SFX6000_SHDLCGuide1.1.pdf, Section 7.2. 

The description for  execution error 0x42 is: "Sensor Busy Error" (The sensor is busy at the moment. Please note that the power-up time after a reset is 300ms)

bposk, 2024-05-28:
Hi @LeonieFierz 

Thanks for that information. Is there any reason the sensor would be "busy" and not make the proper readings? I have not changed the functionality of the exampleUsage code but cannot get rid of this "Execution Error". Is there anything you suggest for solving this issue? 

Considering I am getting a response from the sensor (and not a timeout), I have ruled out this being a hardware or wiring issue, and must be something with the code or a faulty sensor itself.

Note: I have incorporated delays necessary so that power-up time is uninterrupted

LeonieFierz, 2024-05-29:
Hi @bposk 

Do the commands for getting the serial number and setting the set point work without an error? 
You could try to use the method `int16_t readMeasuredValue(float& measuredValue);` getting a single measurement.
If you have no flow applied, the sensor closes the valve at some point to avoid overheating, but this should actually emit another error and only be the case after running a few seconds.

bposk, 2024-05-29:
Hi @LeonieFierz 

The `deviceReset()` commands works and I receive a No Error response, but that's a power cycle and not exactly the same as other commands. 

But the rest of the commands do not work and all respond with  `Execution error, status register: 0x42` (Sensor is busy)
The commands I tested are : 
`int16_t readAveragedMeasuredValue(uint8_t measurements, float& averagedMeasuredValue);`
`int16_t readMeasuredValue(float& measuredValue);`
`int16_t getSerialNumber(int8_t serialNumber[], uint16_t serialNumberSize);`
`int16_t measureTemperature(float& temperature);`
`int16_t setSetpoint(float setpoint);`

As far as the sensor functionality, when power is applied to the sensor the internal valve clicks every second or so (pulses). The sensor does this regardless of whether airflow or no airflow is applied. 

However, it has never reached an 'overheating' condition, the valve on top of the sensor does get hot to the touch, but it never stops actuating itself (most definitely not after running a few seconds). Nor I have I seen any different messages on the Arduino Serial Monitor.

LeonieFierz, 2024-05-30:
Hi @bposk 

Ok, so you actually cannot communicate with the sensor at all. The sensor is most likely in a reboot loop, which will switch the valve and sound like it is actuating. On reason could be insufficient power supply. Do you have the sensor connected to a 24V external power supply?

bposk, 2024-05-30:
@LeonieFierz 

Yes Sensor is connected to 24V external power supply, And I just confirmed with a multimeter the sensor is receiving the proper voltage. 

Well from my point of view, the sensor is communicating back to the Arduino. It is returning error codes, but not anything that I expect it to. It is just not communicating **properly**. When disconnecting the sensor and communication lines are no longer valid, the Arduino sends a  `Error Message: Timeout while reading data` to the Serial Monitor, as expected. 

Is there any way I can manually reset the sensor to get it out of this "reboot" mode (besides sending a reset command in firmware)?

LeonieFierz, 2024-05-30:
Hi @bposk 

Yes, you can communicate with the sensor module, but according your description internal communication to the sensor over I2C itself is not working. 

As you confirm the voltage supply is 24V, it might be that your sensor is defect. Please contact our customer support over https://sensirion.atlassian.net/servicedesk/customer/portal/1/group/1/create/65 to follow up on this.

bposk, 2024-05-30:
@LeonieFierz Thanks for the information and advice. I will contact customer support

closed 2024-05-31 by LeonieFierz